microsoft/Azure-Kinect-Sensor-SDK #1844: Using external mocap system as a trigger for multiple kinects

bs97, 2022-10-05:
Hello, 

I am trying to record data using multiple azure kinects and a mocap system (optitrack) simultaneously for a research study, and therefore require exposures to be synchronised to avoid interference. 

This means I can use the kinects as a trigger or the mocap system which requires this device (https://optitrack.com/accessories/sync-networking/esync-2/). 

This sync device only supports a 3.3V output trigger signal, however, the kinect requires a 5V trigger signal as input (https://learn.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync). Is this 5v limit strict or will it still work?

The second option would be to use the kinects as a trigger, however, as there are multiple kinects (either in a star or daisy configuration), the master device has no output signal as it is being used the the other kinects. In this case, can the sync out port on one of the subordinate devices be used? 

Any help with this matter would be greatly appreciated.

dasparli, 2022-10-10:
The lower threshold for the sync input signal to reliably trigger is Vcc * 0.7 which is 3.5V if Vcc is 5V.

The sync output from one of the subordinates is 5V so if the Optitrack system can't handle this voltage you may need to consider stepping it down first.

bs97, 2022-10-13:
Thanks @dasparli!

closed 2022-10-13 by bs97

meligranda, 2023-05-10:
Hello, @bs97 I have a question. I'm trying to connect the azure kinect with optitrack to desynchronize them to avoid interference but I don't get any signal from the kinect. did you manage to do it? If so, how? Who did you take as master and who as subordinate? 

Thanks!

bs97, 2023-05-10:
@meligranda You must set the kinect as master and optitrack as subordinate. And the kinect must be connected to the optitrack system via ethernet. Make sure you use the isolated output port on the kinect and set the optitrack frame rate to multiply from the kinects default 30hz e.g. x 4.

meligranda, 2023-05-12:
Thanks! I will try

Entongsu, 2024-06-10:
@bs97 I have a question. If the trigger level of my camera system is 1.8V/3.3V/5V, could I still use the optic track as a subordinate?  For the external synchronous input of sync 2, only high impedance and VESA stereo support are available at 3.3V. Thank you!