microsoft/Azure-Kinect-Sensor-SDK #1847: Using synthetic ir images and depth images for body tracking

a969152504, 2022-10-12:
Hi, i am trying to generate synthetic ir images and depth images by using a 3D human model from a scanner and a background which is captured from Kinect, and putting them into k4abt SDK to see if it can find the skeleton of the 3d model.
The value of depth images are calculated by opengl perspective view.
The value of ir images are decided by specular lighting and the depth.
However the results are inaccurate, i am wondering how did you generate the synthetic ir images mentioned in the below Powerpoint?
https://www.microsoft.com/en-us/research/uploads/prod/2020/01/AKBTSDK.pdf

qm13, 2022-10-18:
We implemented a virtual version of the Azure Kinect depth and RGB sensors including noise profile, placed AKDKs in multiple locations in a virtual scene with animated avatars, and ray traced the scene from the perspective of each AKDK.

a969152504, 2022-11-01:
@qm13 I see, is there any source code for it or is it open source?