microsoft/Azure-Kinect-Sensor-SDK #1848: Playback seek_timestamp() and get_next_capture() produce erroneous capture at the end of MKV

Zumra, 2022-10-12:
If we access the MKV with subsequent calls to get_next_capture(), for the last capture in MKV we get valid both color and depth images.
From documentation:
> The first call to [k4a_playback_get_next_capture()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/feature/support-custom-track-record-playback/group___functions_ga32cc6cc2703ab0b113084bd58c731834.html#ga32cc6cc2703ab0b113084bd58c731834) after [k4a_playback_seek_timestamp()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/feature/support-custom-track-record-playback/group___functions_gaea748994a121543bd77f90417cf428f6.html#gaea748994a121543bd77f90417cf428f6) will return the first capture containing an image timestamp greater than or equal to the seek time.

This is true for all the captures in the MKV, but for the last one - returned capture has invalid depth image instead of a valid one which we know exists.

Repro code:
```
auto mkv = k4a::playback::open(mkvPath.string().c_str());
k4a::capture capture;
std::chrono::microseconds colorTimestamp;
std::chrono::microseconds depthTimestamp;

// Obtain the last capture in MKV with both color and depth images valid.
while (mkv.get_next_capture(&capture))
{
  if (capture.get_color_image() && capture.get_depth_image())
  {
    colorTimestamp = capture.get_color_image().get_device_timestamp();
    depthTimestamp = capture.get_depth_image().get_device_timestamp();
  }
}
std::cout << colorTimestamp << ", " << depthTimestamp << std::endl;

// Querying the capture with color image timestamp, returns capture with no corresponding depth image.
mkv.seek_timestamp(colorTimestamp, K4A_PLAYBACK_SEEK_DEVICE_TIME);
mkv.get_next_capture(&capture);
std::cout << capture.get_color_image().get_device_timestamp() << ", " << (capture.get_depth_image() ? "valid" : "invalid depth") << std::endl;

```
Actual result of the above code is:
> 110215777us, 110215755us
> 110215777us, invalid depth

Expected result is:
> 110215777us, 110215755us
> 110215777us, valid

We've identified the issue in one recorded MKV with the following properties:
* frame rate = 5 fps
* record length = 110s
* color_mode = 1080p
* depth_mode = NFOV_UNBINNED
* imu = ON
* exposure_control = -9

SDK version: 1.4.1
Firmware version: color - 1.6.110, depth - 1.6.79

dasparli, 2022-11-03:
There is probably a bug that prevents you from seeking the depth frame when its timestamp is earlier than the color frame. To get around this you can seek the depth timestamp in which case you should be able to find both the depth and color frames.

Zumra, 2022-11-07:
Thanks @dasparli. That's exactly how I'm working around this issue until it gets fixed in the SDK.