microsoft/Azure-Kinect-Sensor-SDK #1849: converting mkv files created by azure kinect with depth to mp4

hadiiahmed, 2022-10-16:
I would like to convert the dataset i created using recordings made by me via azure kinect camera to mp4 without losing any data 
is that possible

rajkundu, 2022-10-29:
It is certainly possible, but there are many ways to do this. One option might be to use VLC Media Player's "Convert/Steam" Functionality, but if you have a large dataset, you may benefit more from writing a script to do this.

dasparli, 2022-11-03:
You can use ffmpeg tool to convert the mkv file to mp4 file through this command 
.\ffmpeg.exe -i <mkvfile> -map 0:0 -c copy out.mp4

hadiiahmed, 2022-11-12:
let me rephrase
I have multiple videos taken by azure kinect DK the mkv files contain RGB DEpth etc.
i would like to extract from each mkv file a depth video in mp4 to use it in my dataset benchmarking
when i tried to use ffmpeg to convert from png to mp4 the depth shows very dark barely visible to the naked eye

dasparli, 2022-11-14:
If you want a mp4 file that displays the colorized depth data the way it's done in the k4aviewer you will need to write this yourself. Fortunately, you can use the code in the k4aviewer utility as a guide on how to do this.

underbelly69, 2024-05-15:
To adjust the white and black levels, you can use the curves filter in FFmpeg. This allows you to remap the brightness values, which can help to emphasize contrast in specific areas, such as making a dark depth map more distinguishable. Here’s how you can do it:

bash

ffmpeg -i input.mkv -vf "curves=preset=lighter" -c:a copy output.mp4

However, if you want more control over the white and black levels, you can use the colorlevels filter. Here's an example:

bash

ffmpeg -i input.mkv -vf "colorlevels=rimin=0.0:gimin=0.0:bimin=0.0:rimax=0.5:gimax=0.5:bimax=0.5" -c:a copy output.mp4

In this command:

    rimin=0.0:gimin=0.0:bimin=0.0: Sets the minimum input value for the red, green, and blue channels to 0.0 (black level).
    rimax=0.5:gimax=0.5:bimax=0.5: Sets the maximum input value for the red, green, and blue channels to 0.5 (white level).

This example effectively increases the brightness of all pixel values, remapping the range [0.0, 0.5] to the full range [0.0, 1.0], which makes dark areas brighter and reveals more details.

You can adjust rimax, gimax, and bimax to different values depending on how much you want to brighten the image. For example, setting rimax=0.7 will remap the range [0.0, 0.7] to [0.0, 1.0], making it slightly less bright than the previous example.