microsoft/Azure-Kinect-Sensor-SDK #1850: Azure Kinect synchronous acquisition does not work

yuzhipeng66, 2022-10-19:
Hello, I'm happy to use Azure Kinect and I want to collect synchronous depth data from 8 Azure Kinect. 
In order to do this, I design a extern synchronous triger generator to create the synchronous signal.
However, in my test, I find that the synchronous triger seems can only control the start of the acquisition. Even I pulling out the 3.5mm synchronous line, the slave camera is still working.
I suspect there is something wrong with my extern synchronous triger generator, so I observe the synchronous output waveform of the Azure Kinect with an oscilloscopesignal, and use a signal generator to create same waveform as extern synchronous triger signal. But the problem is still exists.
does the camera can not capture every frame synchronously or I forgot some thing need to do ?
Your reply will be appreciated.

yuzhipeng66, 2022-10-19:
I follow the guide in https://learn.microsoft.com/zh-cn/azure/kinect-dk/multi-camera-sync
And I noticed that there is a note in this picture.
[image]
It says that "The camera will not start untill it receives a start singal from the master device". Is that means only the first triger signal make sense?

rajkundu, 2023-03-05:
@qm13 @wes-b Just a simple question - why do subordinate cameras continue to stream captures after their sync cable is removed? Are synchronization signals just sent once at the beginning of streaming, or are they sent every time a capture is taken (i.e., 30 Hz @ 30 FPS)?