microsoft/Azure-Kinect-Sensor-SDK #1852: I want to read a offline human mkv video to inference the keypoints and mask of human in this video,but the capture in "tracker.enqueue_capture(capture)" and capture in "k4a_playback_get_next_capture(playback_handle, &capture)" have different classes,how to make them same?

wangcc123-dave, 2022-10-26:
**Describe the bug**
I want to read a offline human mkv video to inference the keypoints and mask of human in this video.
 "k4a_playback_get_next_capture(playback_handle, &capture)" is to get the capture from the offline mkv video,the class of this capture is k4a_capture_t.And "tracker.enqueue_capture(capture)" is to get the inference results(keypoints, mask) of the capture, the class of this capture is k4a::capture.How to make them same?

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version:Windows 10
 - SDK Version:  1.4.1
 - Firmware version:  

**Additional context**

wangcc123-dave, 2022-10-26:
[image]

wangcc123-dave, 2022-10-26:
[image]
This s the description of the capture calss,how to convert k4a::capture calss to k4a_capture_t class?

yuzhipeng66, 2022-11-04:
hi, In my opinion, the [ca](k4a::capture) means a capture, witch include a RGB image, a depth image and so on, and the k4a_capture_t means it is only one image, so you can use k4a_capture_get_color_image() for k4a_capture_get_depth_image() to get the color image or depth image.
I think this web can help you:
https://learn.microsoft.com/zh-cn/azure/kinect-dk/retrieve-images