microsoft/Azure-Kinect-Sensor-SDK #1853: Azure Kinect Sensor SDK for Ubuntu 20.04 with ARM64

ridasaleem0, 2022-10-28:
Do we have an official installation package for Ubuntu 20.04, Arm64 architecture? I cannot seem to configure it on my system. Any leads? @

dasparli, 2022-10-31:
The current SDK only supports Ubuntu 18.04. There are a couple of posts on this board that explain how to add support for 20.04 yourself with varying degrees of success. Just search for '20.04' to find them.

ridasaleem0, 2022-10-31:
I was successfully able to install K4a on jetpack 5.x and Ubuntu 20.04.

closed 2022-10-31 by ridasaleem0

nonmeaning, 2023-05-02:
@ridasaleem0 ,  could you  share your experience of installing  the k4a in jetson with us , many thanks for you

jongyaoY, 2023-08-18:
Here is how I install the SDK on a Jetson Orin with Ubuntu 20.04

```bash
curl -sSL -O https://packages.microsoft.com/config/ubuntu/18.04/packages-microsoft-prod.deb
sudo dpkg -i packages-microsoft-prod.deb
rm packages-microsoft-prod.deb
```
after these steps edit "/etc/apt/sources.list.d/microsoft-prod.list"
```bash
sudo gedit /etc/apt/sources.list.d/microsoft-prod.list
```
Change "[arch=amd64]" into "[arch=arm64]" and replace the repostory path with https://packages.microsoft.com/ubuntu/18.04/multiarch/prod
After editing it should look something like this:
```
deb [arch=arm64] https://packages.microsoft.com/ubuntu/18.04/multiarch/prod bionic main
```
Save & exit then
```bash
sudo apt update
# Then you can finally install them
sudo apt install libk4a1.4 libk4a1.4-dev k4a-tools
```
Connect Azure to Jetson then you can verify the installation with
```bash
k4aviewer # you might need root or copy 'scripts/99-k4a.rules' into '/etc/udev/rules.d/' to avoid being root
```
Here are the references:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md
https://learn.microsoft.com/en-us/linux/packages
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1263#issuecomment-710698591

hchlhwang, 2023-10-11:
Finally, really appreciate your work! @jongyaoY

ChemicalNRG, 2024-05-08:
`wget -q https://packages.microsoft.com/keys/microsoft.asc -O- | sudo apt-key add -`
`echo 'deb [arch=arm64] https://packages.microsoft.com/ubuntu/18.04/multiarch/prod bionic main' | sudo tee /etc/apt/sources.list.d/microsoft-prod.list`
`sudo apt install libk4a1.4 libk4a1.4-dev k4a-tools`

kaninaba94, 2024-06-22:
There is one issue that libsoundio1 cannot somehow not be installed for ARM.. and k4a depends on it