microsoft/Azure-Kinect-Sensor-SDK #1855: I want to read a offline human mkv video to inference the keypoints and mask of human in this video,I can get depth and ir image,but can't get rgb image,which is 0x0000000000000000<null>.How to solve this problem?

wangcc123-dave, 2022-10-31:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

dasparli, 2022-11-08:
Is it possible that the file simply contains no RGB data? You can check this by playing it back with the viewer utility.

amazing-cc, 2022-11-21:
I have the same problem. I am pretty sure it contains RGB data. 
[image]