microsoft/Azure-Kinect-Sensor-SDK #1863: k4a_calibration_2d_to_3d related

myalos, 2022-11-26:
Hello.
In fastpointcloud example, why are not all points(512x512) valid? Is it because there are places in the image sensor that can not receive light?
`
k4a_calibration_2d_to_3d(
                calibration, &p, 1.f, K4A_CALIBRATION_TYPE_DEPTH, K4A_CALIBRATION_TYPE_DEPTH, &ray, &valid);

            if (valid)
            {
                table_data[idx].xy.x = ray.xyz.x;
                table_data[idx].xy.y = ray.xyz.y;
            }
            else
            {
                table_data[idx].xy.x = nanf("");
                table_data[idx].xy.y = nanf("");
            }
`

dasparli, 2022-12-05:
I can confirm that a point can be invalid because the field of view of the sensor is limited and so there is a check to ensure that the unprojected point is not outside the field of view.

myalos, 2022-12-08:
Hi, happy to see you again!  The confirm is what i need,  thank you !  ^ v ^

closed 2022-12-08 by myalos