microsoft/Azure-Kinect-Sensor-SDK #1864: Can anybody help explain undistort example in SDK

myalos, 2022-11-28:
in function compute_xy_range, compute x_min and x_max in 1-mm plane that can project in image sensor
in function create_pinhole_from_xy_range
why fx = 1.f / (x_max_min), ? in my knowledge database, fx = f * \alpha \alpha is the ratio of length in mm to pixel.
why px is -x_min * fx  rather x_min * fx
[image]
i can't figure out how this model works
`ray.xyz.y = ((float)y - pinhole->py) / pinhole->fy;
ray.xyz.x = ((float)x - pinhole->px) / pinhole->fx;
k4a_calibration_3d_to_2d(calibration, &ray, camera, camera, &distorted, &valid);
`
Thanks in advance.

dasparli, 2022-12-02:
[image]

myalos, 2022-12-03:
Very clear explanation, Thank you ! ^v^

closed 2022-12-03 by myalos