microsoft/Azure-Kinect-Sensor-SDK #1865: Azure Kinect stopped working (Invalid Configuration Descriptor)

peter5k, 2022-12-04:
My Azure Kinect was working fine, but recently neither of my two Windows laptops recognize it. The white led is blinking. I am using the same cable and usb port as when it was working, but I also tried every other combination I had available. The Kinect was updated to the latest firmare/SDK when it was working.

USB Device Tree Viewer shows that the hub and the microphone array are ok. But the two cameras are not recognized.

[image]

USB reports:
[Kinect Azure usb port 1.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/10149152/Kinect.Azure.usb.port.1.txt)
[Kinect Azure usb port 2.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/10149153/Kinect.Azure.usb.port.2.txt)

SDK tools are not working, since the cameras are not found.

PC: Dell Pecision 7760 & Asus S510U
OS: Windows 10  Build: 19044.2251

I also plugged the camera into my Steam Deck (handheld PC running Linux based OS), and the devices were identified correctly, so hopefully they're not bricked.
[image]

Any help greatly appreciated!

dasparli, 2022-12-05:
The fact that the same camera and USB cable works when connected to one PC but not another PC is evidence that there is probably an issue with the PC and probably the USB port that you have the camera plugged into. I suggest you try other ports on that PC or even another PC running Windows. You can try connecting a USB 3 device into the USB port to confirm that it works. Also you may want to try supplying external power to the camera using the power USB C cable (with the round DC power jack on the other end) that was supplied with the camera.

peter5k, 2022-12-05:
After trying more cables and combinations I finally managed to find a configuration that works, so everything is ok now. I'm closing this issue.

closed 2022-12-05 by peter5k