microsoft/Azure-Kinect-Sensor-SDK #1886: The whole USB-stack crash during streaming from Azure Kinect

bibigone, 2023-02-10:
Some of our customers experience "USB connection loss" issue while using Azure Kinect devices. As a rule, this USB connection loss looks like a crash of the whole USB-stack in Operation System and in rare cases results in Windows Blue Screen.

**To Reproduce**

This bug is not stable. In some unknown way, it depends on a hardware configuration and a number of connected Azure Kinect devices.

1. As a rule this bug takes place when three Azure Kinect devices are connected (but it can also be reproduced with 2 and 4 devices in use)
2. With some hardware configurations, the sequential unplugging and plugging back to USB port of one Azure Kinect device (before running of k4aviewer app) can catalyze the crash
3. Run the k4aviewer app
4. Wait for a while (sometime it takes just a few minutes before the crash, sometimes it requires hours)
5. k4aviewer reports about connection loss (USB timeout). Actually, the whole USB-stack in OS stops working for a while (for example, other connected USB-devices like Bluetooth adapters also stop working for a few seconds)
6. It looks like that USB-stack automatically restarts in a few seconds, but someone need to replug all Azure Kinect devices to be able to use them.
7. In very rare cases Windows crashes to the Blue Screen

**Logs**
[k4a-crash.log.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/10706054/k4a-crash.log.zip)

**Screenshots**
[image]

**Desktop:**
 - OS Version: Windows 10 Pro 22H2 (build 19045.2486)
 - SDK Version:  [1.4.1](https://download.microsoft.com/download/3/d/6/3d6d9e99-a251-4cf3-8c6a-8e108e960b4b/Azure%20Kinect%20SDK%201.4.1.exe)
 - Firmware version:  [1.6.110080014](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/firmware/AzureKinectDK_Fw_1.6.110080014.bin)
- USB Controller: Intel USB 3.1 eXtensible Host Controller (driver: Microsoft 10.0.19041.2193)

Also, we received reports from our customers about similar behavior under Windows 11 and Ubuntu 18.04.6 Operating Systems.

**Additional context**

- It does not depend on a particular Azure Kinect device unit(s). We tried different ones.
- Similar, it does not depend on USB-cables in use
- Also, it does not directly depend on application that works with Azure Kinect device
- Very often before "crash" Azure Kinect device is receiving damaged color frames (the frequency of damaged frames correlates with the probability of the near USB-stack crash). These damaged color frames can be visible even in any Web-camera application (like "Camera" Windows standard application) and looks as horizontal blocks of squares with noise:

[image]

openmoverse, 2023-02-22:
Hello @bibigone , 
we also are getting such issues with variable number of devices, configurations and we're getting pretty much same logs with the one you've uploaded. Have you managed to overcome this issue in any way?

openmoverse, 2023-03-01:
Hello again. 
So after alot of troubleshooting, we found a way to come around it. First of all, we found an old Laptop of ours with some non-updated Win10 installation there (we were suspicious that our win11 installation had something to do with the system freezing).
On that win10 installation, our software were working properly and when the errors occured, it was not causing system freeze, but still we had errors.
The other thing we did and it helped, is to move 1 specific device, to another (type C) usb controller. It was the device that had its extension USB cable externally powered (again, we were suspicious about it).
With those two changes, we're able to run our software.
Good luck