microsoft/Azure-Kinect-Sensor-SDK #1891: Hi.

patrashu, 2023-03-02:
**Describe the bug**
".\k4arecorder.exe -c 720p_YUY2 -l 1800 output.mkv" is not working properly.

**To Reproduce**
hello. I bought a camera to record video using Azure Kinect. Up until about two months ago, there was no problem in the process of recording video using the camera. However, recently, an error like the picture below has started to occur while filming a video. Due to this, the project I am working on has been delayed and I am very angry. I think it's a problem with the latest windows update. We hope to receive an answer to this problem as soon as possible. thank you

**Expected behavior**
I want to solve this problem

**Logs**
[2023-03-02 17:23:33.448] [error] [t=22060] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\internal\matroska_write.cpp (487): matroska_writer_thread(). Disk write speed is too low, write queue is filling up.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS = Windows 11
- SDK = 1.4.1
- Firmware = 1.6.110079014

**Additional context**

99991, 2023-03-02:
Perhaps your disk write speed is too low? How fast is it? You can measure it with this python script:

```python
# Measure write speed by repeatedly writing 1 MB of zeros to a file.
import os, time
for _ in range(100):
    t = time.perf_counter()
    with open("some_random_file", "ab") as f:
        f.write(b"0" * 1000_000)
    print("%12.3f MByte/sec written to disk" % (1 / (time.perf_counter() - t)))
os.remove("some_random_file")
```

patrashu, 2023-03-07:
Thank you.
After researching for a week, I confirmed that the error I was experiencing was caused by the latest Windows 11 update (KB5022913). So, I'm going to contact the Microsoft team about this issue. If the following error appears after the latest Windows 11 update, it seems that it can be solved by returning to the previous Windows version.