microsoft/Azure-Kinect-Sensor-SDK #1893: Python API for playback of .mkv files

JJLimmm, 2023-03-03:
**Is your feature request related to a problem? Please describe.**
Not a problem but rather a question on whether it is available.

**Describe the solution you'd like**
Opening of a recorded .mkv file using the python wrapper

**Describe alternatives you've considered**

**Additional context**
I want to open up recorded .mkv files for use together with the python wrapper in this SDK

JJLimmm, 2023-03-03:
Similar to the question asked in #1571 , but the topic was sidetracked with other unrelated questions and left unanswered. @qm13

rajkundu, 2023-03-05:
This isn't a direct solution to your answer per se, but I have been using [PyK4A](https://github.com/etiennedub/pyk4a) for working with the Kinect DK in Python, and it has been great for me. It has the ability to read recorded sequences - see the [PyK4APlayback](https://github.com/etiennedub/pyk4a/blob/master/pyk4a/playback.py) class for more.

JJLimmm, 2023-03-06:
> This isn't a direct solution to your answer per se, but I have been using [PyK4A](https://github.com/etiennedub/pyk4a) for working with the Kinect DK in Python, and it has been great for me. It has [...]

Thanks, i was looking at that but with my previous tests and experiments for both the PyK4A repo and another repo (PyKinectAzure), the transformed depth map converted to Depth 3D (Point cloud) map has some discrepancies in depth data as compared to retrieving it with the original SDK's APIs.  

So i was hoping if this original SDK has the python wrapper for the playback function as well similar to those 2 repos.