microsoft/Azure-Kinect-Sensor-SDK #1901: SDK causing ROS2 driver to fail: "Queue_imu was popped in a disabled state"

JeffR1992, 2023-03-09:
**Describe the bug**
I'm using the Azure Kinect SDK with the Azure Kinect ROS2 drivers in Ubuntu 22.04 in Docker. The Azure Kinect is using a depth mode of WFOV, a color resolution of 2160p, and a target framerate of 15 fps.

I'm performing multi-hour ROS2 driver tests for the Azure Kinect to test driver stability over extended use periods, and see that I often get crashes after 4-5 hours of running the camera with the ROS2 drivers. In particular, the ROS2 node crashes with the following error:

[image]

It looks like the error is caused by the Azure Kinect SDK, and in particular there seems to be an issue with IMU data transfer over USB. Are there any reasons why the SDK would cause the ROS2 driver to crash after so many hours?

NOTE: During the tests there are no other tasks running other than the ROS2 Azure Kinect driver and a ROS2 node to log some data, and the only device connected to USB is the Azure Kinect.

joao-corvo, 2023-09-06:
Any updates on this issue?

niccarey, 2023-09-08:
I have this happen intermittently after only a few minutes of using the camera. Very frustrating, frequently can only solve by rebooting the device.