microsoft/Azure-Kinect-Sensor-SDK #1905: I am using the latest SDK with multiple cameras connected and powered by USB C only. All cameras have the latest firmware. I see consistent whole level system hangs (all I can do is reboot) when using serval devices at once - e.g. using multiple instances of the Kinect viewer app or 3rd party solutions. There is nothing in the event log to suggest what happens. What else could I do to investigate the cause of the system lock up? This is on Window 11. The lock up occurs consistently, within a few minutes and normally within a free seconds.

DrMaxP, 2023-03-16:
I still need to look at the Azure Kinect logs, I've enabled the registry keys but can't locate the logs.

But sure if this is a USB issue or a low level driver issue.

DrMaxP, 2023-04-16:
So the system is more stable now, in that I can connect 8 devices and they'll stream from Azure Kinect viewer for much longer than they used to. I've shifted some of the cameras to a 4 port StarTech card which are now connected via USB-A->USB-C + power rather than all being powered by USB-C. It's certainly not reliable - I still see system crashes, but they are not immediate (i.e. within 1 minute). Yesterday they all streamed for >1hr, which they would never do before. 

My conclusion is that the instability was coming from multiple USB stacks running at the same time. 

I have one Kinect that's somewhat unreliable in terms of connecting / errors.

DrMaxP, 2023-04-16:
https://user-images.githubusercontent.com/35763718/232291109-9c3a6874-a3d8-4e3f-956e-3fba8bd954f1.MOV

moversevlad, 2023-04-26:
Hello! For clarification, can you please share the model of the USB hub you're using? We also deal with those kind of problems and we seek to find a solution. Also, do you know at what FPS you're getting data? Are there any framedrops?
Thank you

DrMaxP, 2023-05-12:
>   Hi,
> I have connected it to different USB ports, including those using the Renesas controller recommended by Microsoft. I have 7 other Azure Kinect cameras which do not show the same issues. Sometime the Kinect will not enumerate (appear in the list to select), sometimes it cannot be opened, sometimes it cannot be started, sometimes it will start but show corrupt data, sometimes it will start and then fail after a few seconds / minutes and sometimes it will work fine - e.g. at 30fps. Most of the time it has issues when all other 7 cameras do not.
> Regards,
> Max

Hi, just following this up as I haven't heard anything back to my previous reply.

TomHVR, 2023-09-19:
I think I am encountering a similar issue https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1950

HM281099, 2024-03-14:
how to solve it, I had a similar problem, thanks

TomHVR, 2024-03-14:
What worked for me was to connect the sensors with USB 3.0 extensions or use a 3.0 hub that downgraded the connection to the NUC.

HM281099, 2024-03-15:
thanks for your reply , best wishes @TomHVR