microsoft/Azure-Kinect-Sensor-SDK #1914: Most efficient recording strategy for multiple devices - any solution that records to RAM?

DrMaxP, 2023-04-16:
So, I have a semi-stable system with 8 Azure Kinects. Opening 8 Azure Kinect Viewer instances shows that all are streaming at 30fps (RGB+Depth/IR) - if I switch the viewer to 3D mode, which is less resource intensive. I've started playing with the recorder command line utility, just recording from one device and then playing back the mkv file in the viewer. For **one** device, this shows that it's playing back at around 20fps - which is really surprising. Is that because the recording could only achieve 20fps or is it a playback issue.

The system it's on is pretty high spec - 12900K, 4090 GPU, 64GB RAM, WD Black 850 NVMe SSD - so I'm really surprised by this playback/record speed. 

This leads me to another suggestion - are there any apps that will record to RAM and only write to disc when the recording is finished? That should be about as efficient as you can get and with 128GB of RAM I should get 1-2 minutes even across several devices? If not, I guess I'll need to look into coding this.

ChristopherRemde, 2023-06-14:
Hey! I'm working on a multi-capture solution, and in my experience, storage speed is not the bottle neck here, considering you have an NVME SSD (even a cheap one) in your system. I guess the recorder command line utility is just not very performant.

The capture solution I'm working on saves either pointclouds or raw images (jpg and tiff depth) to disk. Pointcloud saving is pretty expensive, but if you just want to save the raw data, it is very fast. You can try out the latest branch here, but beware that it is still pretty much in development and might have some bugs.

https://github.com/Elite-Volumetric-Capture-Sqad/LiveScan3D/tree/development

DrMaxP, 2023-06-14:
Thanks, I'll definitely take a look.