microsoft/Azure-Kinect-Sensor-SDK #1940: macOS support

PimTournaye, 2023-07-29:
We've been begging, pleading, and damn near groveling for macOS support for Azure Kinect for what feels like an eternity. But what do we get in return? Silence. Absolute, deafening silence. And to top it off, the highest-rated feature request on the feedback forum got shot down without even a hint of an explanation. Seriously?!

How hard can it be? I mean, come on! Microsoft, you're a tech giant with vast resources at your disposal. You roll out updates and features left and right, but somehow, macOS support seems to be a taboo.

The community has been practically screaming for macOS support! There's a massive user base out here craving to get their hands on Azure Kinect, but you're leaving us high and dry. It's like you're deliberately ignoring us, and it's not just frustrating; it's downright disrespectful.

Developers on macOS aren't asking for the moon here. We know you can spare a few devs to tackle this issue, especially since Linux support is already a thing. So, why the delay? It should be an easy port from Linux, right? Are you waiting for the dinosaurs to come back to life before addressing this?

Microsoft, if you care even a smidgen about your developer community, you'll stop stalling and start acting. Give us a straight answer - why is macOS support being pushed aside? If it's a technical challenge, fine, let us know. If it's some weird bureaucratic nonsense, spill the beans!

We're tired of feeling like the stepchild of your development family. Stop hoarding the money and start investing in making Azure Kinect the kick-ass tool it should be - for ALL developers, not just the Windows folks. You've got the power to turn this around and earn some serious respect from your disgruntled macOS devs.

_Thanks for writing this, ChatGPT. My thoughts exactly._

mindbeast, 2023-08-15:
It's clear that microsoft has totally abandoned support for the project. Why would you expect any feature work?

zhonghong322, 2024-01-13:
Due to Microsoft's depth engine library not supporting macOS, Azure Kinect is currently not available for macOS.

zhonghong322, 2024-01-13:
1、Femto Mega (Good news, good news!)
Femto Mega is a programmable multi-mode Depth and RGB camera with real-time streaming of processed images over ethernet or USB connections. The camera uses Microsoft’s industry proven ToF technology and the NVIDIA® Jetson™ platform to deliver a comprehensive Depth and RGB vision platform for computer vision and AI developers. With a wide field of view (FOV) and high resolution depth sensor, it’s ideal for robotics, manufacturing, logistics and healthtech applications.

https://www.orbbec.com/femto-mega-document/

Compared to Azure Kinect, it has the following advantages
* Lower CPU usage.
* Network functionality support.
* macOS support. （MacOS will be supported in this month's version of the Orbbec SDK）

2、GitHub Link

https://github.com/orbbec