microsoft/Azure-Kinect-Sensor-SDK #1960: Open K4A Device failed! with the set up following the instructions

anyRandomHuman, 2023-11-21:
While trying to run exampleRecording.py from [pyKinectAzure](https://github.com/ibaiGorordo/pyKinectAzure), following errors are reported:

[2023-11-21 13:06:35.820] [critical] [t=22837] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2023-11-21 13:06:35.820] [error] [t=22837] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2023-11-21 13:06:35.820] [error] [t=22837] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
Open K4A Device failed!
[image]

I have seen similar issues, but they are solved by setting up following [instructions](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup) . but i can actually find the setup file with correct content in the correct folder.
[image]

I don't if some other .so file affected this and i also have no clue how to check that.

**Desktop:**
 - OS: Ubuntu 20.04
 - SDK Version: 1.4.1

TangBoheng, 2023-11-29:
I get the same error, have you solved this problem?

anyRandomHuman, 2023-11-29:
> I get the same error, have you solved this problem?

I have found the problem. The camera was unplugged. So be sure to first check that the your configuration was correct following the instructions(see above) and also check if it is powered.

closed 2023-11-29 by anyRandomHuman